meshtastic/protobufs #47: lint our protobufs?

mkinney, 2021-11-30:
Should we use a linter of sorts for our protobufs?

See https://yoheimuta.medium.com/how-to-lint-your-protocol-buffer-files-very-quickly-and-easily-cf2d228a8fc6

mc-hamster, 2021-11-30:
Yes!

mkinney, 2021-11-30:
I have a branch, but I cannot tell if our standards are 2 spaces or 4 spaces of indentation.

sachaw, 2021-11-30:
I looked at this a while ago, the main issue is auto-formatting to the comment structure that we use:
```
/*
 * Comment
 */
 ```
 @mkinney if your able to find something that supports this, that would be great!

mkinney, 2021-11-30:
I was able to run with 2 spaces to get everything consistent. But, I had to go back and fix the comments to match with our style. The linter does not complain about the spacing of the newly fixed comments. So, I think we're good.

The bottom line is that the linter is not that smart (or I'm not that smart to config it) to fully check our comments. However, it will be checking for other things, and that is good.

@sachaw Let me know what you think.

sachaw, 2021-11-30:
Good start, as long as it doesn't mess with the comments, something is better than nothing.

mkinney, 2021-11-30:
If you like, then we can create another issue to add linting to a github action

Also be sure to checkout: https://github.com/plexsystems/vscode-protolint

mkinney, 2021-11-30:
We could use this for the github action: https://github.com/plexsystems/protolint-action

sachaw, 2021-11-30:
Yeah, seems fine

PR #48 (2021-11-30): Add lint
Fix #47
closed 2021-11-30 by mc-hamster