meshtastic/protobufs #107: Rename and repurpose environmental plugin

sachaw, 2022-02-26:
### Proposal:

Repurpose the existing environmental plugin to be a generic telemetry module.

The functionality should include:

1. Allow rapid streaming of `AirTime`, `BatteryLevel`, `ChannelUtil` and other high frequency data over the phone api, likely to be own message:

``` proto
message deviceMetric  {

  float channel_utilization = 1;

  float air_util_tx = 2;

  int32 battery_level = 3;
}
```
2. Allow generic sensor data to be sent over the mesh (current functionality)
3. For the mesh traffic, have a flagged enum field (in radioConfig.telemetry) to determine what sensor data is sent (identical to the existing `position_flags` field), another generic setting to determine the interval

jfirwin, 2022-02-27:
Will there need to be any consideration to multiple sensors attached to a device at once along with intervals unique to each/any sensors?

sachaw, 2022-02-27:
The interval should be for how often the messages are sent over the mesh, there should be some sort of averaging as we discussed on the call.

garthvh, 2022-06-12:
Done!

closed 2022-06-12 by garthvh