meshtastic/protobufs #165: Add Speed & Heading to GPS data

djscoutmaster, 2022-06-12:


garthvh, 2022-06-12:
Now is the time to add these to the position flags if we want this data

djscoutmaster, 2022-06-12:
Great! Do you need more details? How can i help?

garthvh, 2022-06-12:
I think we need two values here to start https://github.com/meshtastic/Meshtastic-protobufs/blob/e5b5adc196d3593ab15c04101093443ab8b36c9c/config.proto#L93

djscoutmaster, 2022-06-12:
Ive never been able to even successfully build the device project. I won't be much help unless i pay another developer to do it for me :)

garthvh, 2022-06-14:
> Ive never been able to even successfully build the device project. I won't be much help unless i pay another developer to do it for me :)

I am sure this can be arranged.

garthvh, 2022-06-20:
Merged into the protobufs

closed 2022-06-20 by garthvh

djscoutmaster, 2022-09-25:
This got into protobus but did it make it all the way into the app?

garthvh, 2022-09-25:
You can turn the flags all on in the iOS app currently, there is one of the flags that is not working on device and I still need to finish adding the enabled flags to the position log view.