meshtastic/protobufs #283: Protobuf Compilation not Working after Namespacing

ajmcquilkin, 2023-01-19:
Super excited about the namespace change! One issue -- I'm not able to compile files with the new import style of `import "meshtastic/FILE.proto"`. I'm pointing protoc towards the new `protobufs/meshtastic` directory, but it will only compile when I replace all `import "meshtastic/FILE.proto"` imports with `import "FILE.proto"`. Not super familiar with protobuf compilation so very possible I'm doing something wrong.

caveman99, 2023-01-20:
look in the firmware repository bin folder. the files 'regen-protos' have working protoc calls, assuming nanopb is setup right. I had to tweak the windows version a bit, path inclusion and expansion is not a 1:1 match across every build platform.

Also the sed commands that follow pretty much remove the namespacing from the importfiles again, otherweise there would have been a heap of changes to the sourcecode.

ajmcquilkin, 2023-01-21:
Following up on this, this was a mistake on my end. I had the compiler pointing to the `/protobufs/meshtastic` subdirectory instead of the `/protobufs` directory, which was causing the import errors. Closing as a non-issue 🤦‍♂️

closed 2023-01-21 by ajmcquilkin