meshtastic/protobufs #325: Add Telemetry Type: Compass

MisterRager, 2023-03-01:
In order to support dynamic heading from the compass to supplement/replace the derived heading from GPS, it would help to add a telemetry type for Compass sensors. These sensors all at least provide their raw 3-axis measurements. Some also will provide a derived 2d azimuth in the x/y plane that can be used after calibration directly.

caveman99, 2023-03-01:
The compass sensors are not a good fit for remote telemetry and transmitting the raw readings. Intended use is to use them on the same device to replace heading information in position packets if the node is stationary or low speed. also rotate the display compass so it's actually pointing to the other device.

MisterRager, 2023-03-01:
Ok - I'll use separate system.

closed 2023-03-01 by MisterRager