meshtastic/protobufs #395: [breaking] move `uplink_enabled` & `downlink_enabled` from `Channel` to `MQTT` module config

andrekir, 2023-10-12:
https://github.com/meshtastic/protobufs/blob/master/meshtastic/channel.proto#L75-L84

There is no need to have `uplink_enabled` and `downlink_enabled` as part of `Channel` and `ChannelSet` since they don't need to be shared with all nodes in a local mesh to be able to communicate.

Instead, it would be better to have the channels selection for uplink and downlink moved into the MQTT module configuration.

thebentern, 2023-10-13:
Seems like a good item for the 3.0 project board

todd2982, 2024-11-22:
Will this not prevent me from avoiding downloads from the internet from the public mesh, while still allowing my private channel to be shared?

I currently have uplink enabled, and downlink disabled for the public channel, and both enabled for a private channel.  Will this remove that capability?