meshtastic/protobufs #412: [Feature Request]: Add Philippines (PH) - AS923-3 (915-918 Mhz) LoRa Region Setting

markbirss, 2023-12-16:
### Platform

ESP32

### Description

There is currently no region setting for Philippines (PH) using AS923-3 (915-918 Mhz, 25mW)

https://www.reddit.com/r/HeliumNetwork/comments/m8ljdd/anyone_running_a_hotspot_in_the_philippines/
The 434/868 Mhz frequencies are not free
[image]

https://lora-alliance.org/wp-content/uploads/2021/05/RP-2-1.0.3.pdf
Page 16 of 94 

[image]

j3ffrw, 2024-01-16:
Hi @markbirss,

  Good day does [MC-03-06-2017.pdf](https://github.com/meshtastic/firmware/files/12506777/MC-03-06-2017.pdf) do supersede [MC-03-08-2013](https://region7.ntc.gov.ph/images/LawsRulesAndRegulations/MC/WDN/MC_03-08-2013.pdf)?  because it only mentioned "Non-specific SRD,telemetry ... etc" while 915-918MHz mentioned in MC_09-09-2003 in is under H(I/Y)PERLAN/RLAN or Wireless data networks. [MC-01-02-2013](https://region7.ntc.gov.ph/images/LawsRulesAndRegulations/MC/LowPowerEquipment/MC_01-02-2013.pdf). 
  I might have misunderstand how this MC amends works though if this a union or difference. I am interested in getting the frequency right before purchasing my first meshtastic device. 

Regards,

Jeff

markbirss, 2024-01-16:
@j3ffrw 

if you are based in PH it could/would help if you could follow up with the relevant bodies mentioned in the documents, I only visited Nov 2023 and hoped to have some finality then but could not get it.

The legal Helium use according to these documents has not been published everywhere else

j3ffrw, 2024-01-17:
hi @markbirss ,

   Thanks for the advice and tips on the device, yes I'm in PH let me see what I can do to get a formal stance of NTC on AS932-3.
 
Regards,
Jeff