meshtastic/protobufs #424: Delete File by path admin message

garthvh, 2024-01-24:
Years later there remains no way to delete the CSV file for range tests, create an admin messages so this file can be deleted.

geeksville, 2024-01-24:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/how-to-reset-the-csv-file/9430/5

andrekir, 2024-01-25:
this still works:

```
powershell: Invoke-RestMethod -UseBasicParsing -Uri "http://meshtastic.local/json/fs/delete/static?delete=static/rangetest.csv" -Method Delete
```

maybe one of the web devs can implement this on the web client.

thebentern, 2024-01-28:
We can probably re-use that underlying method. Adding an admin message will let us do it on non-wifi hardware like Pico and Nrf52.

PR #425 (2024-01-29): Delete files admin message
Closes #424
closed 2024-01-29 by thebentern