meshtastic/protobufs #455: How to extract message id directly from protobuff?

lollo03, 2024-02-28:
Hi, I am writing a small " router " application for meshtastic mqtt messages. I need to extract the message ID from the raw message that the nodes send to the mqtt broker. However, I would like to not use the meshtastic library or any protobuff (I am writing my application in go)

There is some way to extract the message ID directly from the raw data without using protobuff libraries?

I tried to search on the docs but it appears that there isn’t any indication on how the data are encoded.

I also tried to look at some data with a hex editor and I wasn’t able to find the message id.

Can anyone help me?

Best regards, Lorenzo

garthvh, 2024-02-28:
There is not

garthvh, 2024-02-28:
Google makes protobufs, there is a good go library.

closed 2024-02-28 by garthvh

lollo03, 2024-02-28:
Thank you for your answer!