meshtastic/protobufs #556: Re-using Tagged Versions

djekl, 2024-08-14:
Hi,

Can this project please not re-release a tagged release. This is becoming more and more common as of late.

I have noticed since I created my PHP version of these protobufs that version numbers are being re-tagged for release multiple times.

For example, yesterday version `v2.4.3` was released, and then this version number was again released just now. This means that I need to manually delete the tag and release from my repo, and finally roll back any commits between releases. This is so that the github automation can re-run to update the code to the latest release.

Please find attached a screenshot of re-used tagged releases from this repo.

[image]

thebentern, 2024-08-14:
Maybe you should on the firmware release tag. Nothing happens until that is available anyway.
Also, if you look at the one we just re-tagged, it was line formatting. No actual protobuf changes were.

closed 2024-08-14 by thebentern

djekl, 2024-08-14:
Maybe don't tag a release until its finalised? I am trying to support this with a PHP library for the Meshtastic protobufs.

If SemVer were followed, then you could use pre-releases and that would resolve the issue.

The automation watches only one repo, it shouldn't need to look at a separate repo to know when a tagged release is OK to use...

AeroXuk, 2024-08-14:
It’s a shame tags and version numbers get so casually reused across the Meshtastic repos. It makes the project look amateur when there are actually many hard working people behind it.

I really hope the Meshtastic repos can become better disciplined with the use of tags and version numbers.

djekl, 2024-08-14:
@AeroXuk thanks for the support. Now this is closed they won’t see it. Which is another thing they do. So far my perspective from the team is that of hostility. Tickets are closed far too early, and only their opinion seems to matter. Its like this is open source only in name, not in culture unfortunately

thebentern, 2024-08-14:
> @AeroXuk thanks for the support. Now this is closed they won’t see it. Which is another thing they do. So far my perspective from the team is that of hostility. Tickets are closed far too early, and [...]

I'm sorry you felt there was hostility, but community usage of the protobuf versions are not something were even on our radar. We can only keep track of consumers in the official firmware, apps and clients. We typically produce an alpha preview release of the firmware for folks to test on the discord before the apps consume the updated protos, so this has not been a problem in the past. 

FWIW I've used semantic versioning in many other projects and am not opposed to adopting it in Meshtastic, but currently things are setup this way across all of the repos and I don't currently have the bandwidth to re-work the CI/CD scripts across it all.