meshtastic/protobufs #586: [Feature Request]: Add a Node/Node user status to one of the periodic broadcast packets

Nestpebble, 2024-09-23:
### Platform

Cross-Platform

### Description

Nodes would benefit from a "status" in the same way that other chat platforms have a user status, e.g. Available, Away, Busy, etc.

However, Meshtastic nodes can be in many more states than just the user, e.g. Currently Unattended, Totally Unattended, Autoresponder only, Answering Machine only, Store & Forward Server, etc.

Adding a suitable set of statuses to a node would also improve the out-of-the-box experience for new users, who would be able to see that although there is a mesh around them, none are attended at the moment.

As more complex status messages are involved, bandwidth consumption increases and so do basic concerns like security and passive surveillance. However, with a few carefully selected statuses, I think this might be a valuable addition to the overall system.

garthvh, 2025-05-14:
There is some good ideas from @GUVWAF in this pull request as well https://github.com/meshtastic/protobufs/pull/667, users are adding some creative emoji status to the user long name we should look at integrating as well and I like @caveman99's idea to use a bitfield for multiple status.

The is_unmessagable has been merged so we don't need 📵 emoji anymore.

NomDeTom, 2025-05-27:
I think this still has some legs left with regards to afk/other statuses. I'll keep it open for now.

danditomaso, 2026-01-05:
When you say periodic broadcast packets, do you mean when node info packets are sent around or do you have a concrete idea of what packet would carry this info?