meshtastic/protobufs #647: MSVC 2022 compile error

hexabyte23, 2025-01-19:
Hi all,

Trying to compile protobuf classes with MSVC 2022 Community, get this :

..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1299): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1300): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1301): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1302): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1303): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1304): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1305): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1306): error C2238: unexpected token(s) preceding ';'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1307): error C2143: syntax error: missing ')' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1307): error C2143: syntax error: missing ';' before 'constant'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1307): error C2737: 'meshtastic::ModuleConfig_SerialConfig::DWORD': constexpr object must be initialized
[... cut ...]
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1309): error C2789: 'meshtastic::ModuleConfig_SerialConfig::DWORD': an object of const-qualified type must be initialized
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1309): note: see declaration of 'meshtastic::ModuleConfig_SerialConfig::DWORD'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1309): error C2059: syntax error: ')'
..\..\thirdparty\Meshtastic\generated\meshtastic/module_config.pb.h(1309): error C2238: unexpected token(s) preceding ';'

Missing some compiler flags ?

Thanks

danditomaso, 2025-01-19:
Can I ask how you went about compiling the protobufs for C#? Or was there a nugget package you used?

hexabyte23, 2025-01-19:
Its C++ files generated by regular protoc tool, no C# here.

vzvezda, 2026-04-05:
I have got the same issue, the reason is naming conflict with `#include "windows.h"`, some of the enums has names that conflicts with defines in "WinBase.h" like this:

`#define BAUD_110          ((DWORD)0x00000002)`

I plan to solve this by not including windows.h and meshtastic protobufs headers in the same compilation unit.