meshtastic/protobufs #648: GA "Push commit to schema registry" is failing since october

KomelT, 2025-01-20:
GA "Push commit to schema registry" has been failing since October and new protobuf versions aren't published.

[image]

sachaw, 2025-01-21:
We can stop publishing to bsr as we aren't using any code generated from there anymore.

KomelT, 2025-01-21:
Oh, I see. I thought Web UI was using protobuf definitions from there, but it inherits them from @meshtastic/js. The JS library gets protobuf definitions from https://jsr.io/@meshtastic/protobufs.

I hope I am not wrong again.

KomelT, 2025-01-23:
Ha, interesting. The JSR looks like it is platform-specific (JS and TS). Buf supports Swift, ts, js, python, go, c, rust, scala, and more. Isn't this like a downgrade? I took a look and saw that .npmrc contains only one line. My friend likes to do custom stuff with Meshtastic and GO; after that, he would need to build custom protobuf.

KomelT, 2025-01-24:
In any way. The current version of protobuf on JSR is 2.5.9. The latest protobuf version in the repository is 2.5.19. It might be time to update so we can add new features to the web UI.

wz2b, 2025-04-13:
I am using this buf registry so it would be extraordinarily helpful to me if this would be fixed.

wz2b, 2025-04-13:
> We can stop publishing to bsr as we aren't using any code generated from there anymore.

You aren't, but others (like me) are still using it.  It looks like it's an easy fix, there's some secret that just needs to be re-configured in CI

r41d, 2025-04-25:
> We can stop publishing to bsr as we aren't using any code generated from there anymore.

Regardless of that I think it would still serve as a very good source of documentation since it's nicer to use than looking at the files directly. Also it's a bit misleading that it's linked in the README without any warning that it contains outdated documentation 🤔

chihosin, 2025-06-08:
This issue confused me as well. I was about to start developing new features for the Web Client, but I couldn’t find the latest Protobuf TypeScript build—until I discovered the #663