meshtastic/protobufs #695: Convert to LGPL to allow more "things" to use meshtastic

mikecarper, 2025-05-27:


garthvh, 2025-05-27:
GPL V3 leaves a nice level playing field.

closed 2025-05-27 by garthvh

ajvpot, 2025-05-27:
Example case:
 
I am considering developing a companion radio that bridges packets from the Meshtastic network to a proprietary network ([Amazon Sidewalk](https://docs.sidewalk.amazon/)) using the PROTO mode on the Serial port. Packets would ingress over Meshtastic, be forwarded to the Sidewalk radio over Serial, be processed by AWS, and exit via another Sidewalk/Meshtastic combo node, similar to how MQTT currently functions. I have taken additional considerations to make sure this would play well with the mesh but I will omit them here because they are not relevant to this issue.

Operation of the companion radio would require a Meshtastic radio running the GPL firmware. I am personally OK with releasing the companion radio software under the GPL, but the proprietary network has an SDK license that prohibits 
> use [] with any software or other materials that are subject to licenses or restrictions (e.g., open source software licenses) that, when combined with [the sdk], would require you or us to disclose, license, distribute, or otherwise make all or any part of such [sdk] available to anyone

A generated API client based on the Protobufs would be GPL licensed, which would require me to license all of the code for the companion radio as GPL. This conflicts with the proprietary license, and effectively makes the project impossible.

I don't believe the Core should be relicensed, just the Protobufs. Relicensing the Protobufs would allow more freedom to create "Meshtastic Compatible" devices, while keeping the spirit of the GPL in that any changes to the Meshtastic Core like routing improvements or additional features must be open sourced.

ajvpot, 2025-05-27:
@garthvh any comment on the above or an alternate solution? Or am I just not meant to do this?

mikecarper, 2025-05-27:
Closing this out will limit the usage of meshtastic with no benefit and only downsides.

garthvh, 2025-05-27:
@mikecarper i deleted the comment on accident, but there are a ton of upsides of the GPL V3

garthvh, 2025-05-27:
> yeah I'm a huge fan of the GPL. In some cases though (this being one of them) a more open license makes sense. ChatGPT says otherwise in this case
> 
> [image]

ChatGPT would say that as MIT is very advantageous to large companies who want to use open source and not contribute back. Meshtastic is the defacto standard already so we can afford to take this risk without letting the fox into the henhouse.

garthvh, 2025-05-27:
> how about a LGPL exception then if making the protobufs open isn't an option?

They are open, and your code would need to be open too which is totally fair. I reject the phrasing here.

mikecarper, 2025-05-27:
https://github.com/meshtastic/protobufs/blob/master/meshtastic/config.proto there's no secret sauce in here to protect. Open to outside development is what I'm after

mikecarper, 2025-05-27:
The glue code for amazon would be FOSS. You can make exceptions stating this is a requirement if using the protobufs without the viral GPL issue that amazon is trying to avoid.

ajvpot, 2025-05-27:
Thanks for your response Garth. I appreciate you engaging in this conversation. Licensing can be contentious and I respect your role as a leader of the project to make these decisions.

I'd like to reiterate that I think the GPL license is appropriate for the Meshtastic Core. However, licensing the Protobufs as GPL leads to a situation that makes it harder for both hobbyists and businesses to work around. It's currently not possible to make a Meshtastic client app that contains any proprietary code.

Even with a hypothetical MIT license on the API definitions, if a business wanted to add additional functionality to Meshtastic they would be required to open source the implementation under the GPL, or recreate Meshtastic Core in its entirety based on the API definition. I believe the level of effort required to not contribute would be the same as developing their own system from the ground up, so they may as well just contribute. This results in a synergistic relationship where contributions are made to Meshtastic from business users, with a low risk of them "co-opting" the project.

I'm just a hobbyist, I do not plan to commercialize this project at this time. Still, the license restricts me from attempting this project. 

Thanks again for your consideration.

garthvh, 2025-05-27:
> https://github.com/meshtastic/protobufs/blob/master/meshtastic/config.proto there's no secret sauce in here to protect. Open to outside development is what I'm after

Proprietary development and outside development are not the same thing. There is plenty of outside development, everyone has to release their code, which is fair.

garthvh, 2025-05-27:
Our whole program is GPL v3, every repo.

mikecarper, 2025-05-30:
```bash
#!/usr/bin/env bash
set -euo pipefail

export GITHUB_USER=
export GITHUB_TOKEN=

if [[ ! -f commits.json ]]; then
  page=1
  first=true
  while :; do
    resp=$(curl -s -H "Accept: application/vnd.github.cloak-preview" \
        -u "$GITHUB_USER:$GITHUB_TOKEN" \
        "https://api.github.com/search/commits?q=meshtastic/protobufs+-org:meshtastic&per_page=100&page=$page")
    items=$(jq '.items' <<<"$resp")
    # Stop once there are no more results
    if [[ $(jq 'length' <<<"$items") -eq 0 ]]; then
      break
    fi
    # Emit each commit object, prefixing commas on subsequent pages
    jq -c '.items[]' <<<"$resp" | while read -r obj; do
      if $first; then
        echo "  $obj" >> commits.json
        first=false
      else
        echo " ,$obj" >> commits.json
      fi
    done
    ((page++))
  done
else
  echo "commits.json already exists; skipping download."
fi

( echo '['
  sed '$! s/$/,/' commits.json   # append commas to every line except the last
  echo ']' 
) > commits_array.json

# Extract each repo full_name from the top-level array
jq -r '.[].repository.full_name' commits_array.json | sort -u > repos.txt

mkdir -p licenses
while read -r repo; do
  license_file="licenses/${repo//\//_}.json"

  # only fetch if we haven't already
  if [[ ! -e "$license_file" ]]; then
    printf "\rFetching license for %s…" "$repo" >/dev/tty

    http_code=$(
      curl -s -w '%{http_code}' \
           -H "Accept: application/vnd.github.v3+json" \
           -u "$GITHUB_USER:$GITHUB_TOKEN" \
           "https://api.github.com/repos/$repo/license" \
           -o "$license_file"
    )

    if [[ "$http_code" == "404" ]]; then
      # no license file → record NONE
      echo '{"license":null}' > "$license_file"
    fi
  else
    printf "\rSkipping %s (already exists)" "$repo" >/dev/tty
  fi
done < repos.txt
echo ""

# extract spdx_id (or “null”)
jq -r '.license.spdx_id // "NONE"' licenses/*.json > all_spdx.txt

# list repos that are NOT GPL/LGPL/AGPL, including “NONE”
paste -d' ' <(ls licenses/*.json | sed 's|licenses/\(.*\)\.json|\1|') all_spdx.txt \
  | awk '$2 !~ /^(GPL|LGPL|AGPL)/ { print $1 " has license: " $2 }'

```

outputs

0dayInc_meshtastic has license: MIT
0dayInc_pwn has license: MIT
bentoml_plugins has license: Apache-2.0
bettio_pocketOS has license: Apache-2.0
bmwiedemann_openSUSE has license: NONE
colonelpanichacks_deepwoods_device_detection has license: NONE
crypto-smoke_meshtastic-go has license: MIT
danpeig_node-red-meshtastic-msg has license: MIT
FreeTAKTeam_FreeTAKHub_meshtastic has license: EPL-2.0
ketan_paragliding-meshmap has license: MIT
knl_dotskel has license: NONE
ldaponte_Meshtastic-Related has license: NONE
liamcottle_meshtastic-map has license: MIT
lisp-mirror_lqml has license: NOASSERTION
lmatte7_goMesh has license: MIT
mannkind_meshtastic2mqtt has license: MIT
meshnet-gophers_meshtastic-go has license: MIT
mkinney_ble-objc has license: NONE
morria_HopsApp has license: NONE
Nate-Boyette_MT-Tkserver has license: NONE
nitroxgas_ESP32-Pluviometer-Temp-Humidity-Meshtastic has license: NONE
patrikcze_meshtastic_go has license: NONE
paulmandal_atak-forwarder has license: MIT
pnm-stockly_my-registry has license: NONE
resiliencetheatre_rpi4edgemapdisplay has license: NONE
resiliencetheatre_rpi4edgemap has license: NONE
rooster-ninja_node-red-farm-test has license: NONE
rosmo_armassi has license: MIT
technologiestiftung_kiezbox-gateway-service has license: NONE
technologiestiftung_kiezbox-meshtastic-python has license: NONE
wel97459_Meshtastic-C has license: Apache-2.0

github search isn't perfect as it missed this as well
https://github.com/jurriaan/meshstellar

mikecarper, 2025-06-15:
Would any Open Source exemption be a possibility here?