meshtastic/protobufs #712: GPL 3 Violations

mikecarper, 2025-06-24:
No License Listed:

- https://github.com/distinctm1nd/2025_utahtastic_bsides_badge/tree/functional/software/protobufs/meshtastic
- https://github.com/meshtastic/python
- https://github.com/nisfeb/mqtt-pipeline/tree/master/mqtt_pipeline/middleware/meshtastic/protobufs/meshtastic
- https://github.com/rjmcdougall/BB/tree/master/bb/src/main/proto/meshtastic
- https://github.com/technologiestiftung/kiezbox-meshtastic-python/tree/kiezbox-2.5.10/meshtastic/protobuf
- https://github.com/whereiskurt/defcon.run.33/tree/main/apps/mqtt/nginx/meshmap/meshmap.net/internal/meshtastic/generated
- https://github.com/Xinyuan-LilyGO/LilyGO-Mini-E-paper-S3/tree/main/Meshtastic/protobufs/meshtastic
- https://github.com/morria/HopsApp
- https://github.com/nitroxgas/ESP32-Pluviometer-Temp-Humidity-Meshtastic/blob/main/include/meshtastic-protobuf.h

MIT:

- https://github.com/daviesgeek/meshtastic-mqtt-discord-bridge/tree/main/src/protos/meshtastic
- https://github.com/dpup/meshstream/tree/main/proto/meshtastic
- https://github.com/FriendlyDev/Meshtastic-PHP-Protobufs/
- https://github.com/ketan/paragliding-meshmap/tree/main/src/protos/meshtastic/meshtastic
- https://github.com/laneboyerre/meshtastic_chat_desktop/tree/master/venv/Lib/site-packages/meshtastic/protobuf
- https://github.com/crypto-smoke/meshtastic-go
- https://github.com/lmatte7/goMesh/tree/main/github.com/meshtastic/gomeshproto
- https://github.com/meshnet-gophers/meshtastic-go
- https://github.com/paulmandal/atak-forwarder/tree/main/app/src/main

Apache 2.0

 - https://github.com/wel97459/Meshtastic-C/tree/main/extern

garthvh, 2025-06-24:
There is a gpl v3 license in the python repo, have you checked these or are you doing an automated only check?

mikecarper, 2025-06-24:
I was looking here as most other projects have the license listed there 

[image]

garthvh, 2025-06-24:
Need to do a more careful check if you are compiling a list of violations, seems like the same approach as before, what is your goal here knowing that a relicensing to MIT is not an option?

closed 2025-06-24 by garthvh

garthvh, 2025-06-24:
This is not the approach we are going with, we are about a mile apart and I would be happy to get together to have coffee and chat, but this approach is not going to work, and we are not re-licensing to MIT.

mikecarper, 2025-06-24:
Enforce the gpl, that's all I'm asking for here

NomDeTom, 2025-06-24:
@recrof I'm not sure if you're aware of the previous issue that was raised [here](https://github.com/meshtastic/protobufs/issues/695). Reading between the lines, I think Mike and Garth discussing this AFK might be the best thing for everyone.