meshtastic/protobufs #740: DeviceOnly is not found in https://jsr.io/@meshtastic/protobufs/doc

fifieldt, 2025-07-22:
The DeviceOnly protobuf is needed by the OEM generator on the website.

For some reason, it's not generated/not found at:

https://jsr.io/@meshtastic/protobufs/doc

Related: github.com/meshtastic/meshtastic/pull/1946

garthvh, 2025-07-22:
It is kind of a weird combo of other messages, does the generator require some fields?

fifieldt, 2025-07-22:
Yeah, the OEM generator needs things like Font size ... let me get the list ...

fifieldt, 2025-07-22:
```
Protobuf.ScreenFonts 
Protobuf.DeviceOnly.ScreenFonts.FONT_MEDIUM,

 21       new Protobuf.DeviceOnly.OEMStore({
 22         oemAesKey,
 23         oemFont,
 24         oemIconBits,
 25         oemIconHeight,
 26         oemIconWidth,
 27         oemText,

```

fifieldt, 2025-07-23:
Awesome! Let's close this then

closed 2025-07-23 by fifieldt